daolis/ioBroker.sun2000-modbus #7: Initial version

daolis, 2023-08-28:
@christi-ko, @RoHa78,
Ich habe ein paar Änderungen gepushed, die das Slave device busy problem fixen sollten.

Den connection issue wenn noch keine/falsche ip eingetragen ist, werden ich später angehen.

Es gibt jetzt 2 intervalle. Es kann aber sein dass ein Durchlauf des Intervalls ausgelassen wird, wenn ein anderer noch läuft. Sie werden jetzt mehr oder weniger hintereinander ausgeführt...

Installiert einfach die version von GIT drüber, ohne die alte zu deinstallieren. Die version hat sich nocht nicht geändert.
Ich werde dann mal ein release machen wenn es einigermassen stable ist.

christi-ko, 2023-08-28:
Ich hab mal getestet und Dir gleich ne Issue 8 hinterlassen, Sorry dafür ...

Was mich verwundert hat war die Log Anzeige, dass ein Update von 3 Werten wirklich so lange dauert. Beim iobroker Modbus Adapter welcher ich auch am Laufen habe braucht er für deutlich mehr viel kürzer.

christi-ko, 2023-08-29:
Welche Library verwendet du für den Adapter? Ist das die gleiche wie bei NodeRed, was wohl dieser ist? [https://www.npmjs.com/package/modbus-serial](https://www.npmjs.com/package/modbus-serial)

christi-ko, 2023-09-01:
Bezüglich der Geschwindigkeit könnte man für manche Register mehrere auf einmal einlesen. So könnte man z.B. bei zusammenhängenden ein wenig Zeit rausholen.
Ich glaube für NPM Modbus gibt es sonst nicht viele weitere Bibliotheken, ein paar gibt's aber ob die besser sind?

daolis, 2023-09-06:
Hab noch jeweils 2 states hinzugefügt, die von storage.chargeDischargePower & grid.activePower berechnet werden.

* storage.chargePower (Mit wieviel Watt wird gerade geladen)
* storage.dischargePower (Wievel Watt werden aus dem Akku geholt)
* grid.feedIn (Wieviel wird eingespeist)
* grid.supplyFrom (Wieviel wird aus dem Netz geholt)

Ist dann z.B. einfacher einen Grafana Chart zu basteln.

RoHa78, 2023-09-06:
Die zusätzliche Werte sind zur Visualisierung super hilfreich, wäre hier eventuell die Leistung der einzelnen Stränge (PV1, PV2) auch sinnvoll? Wäre ja nur das  Produkt aus Strom und Spannung...

gold2605, 2023-10-10:
Hallo, ja der aktuelle Hausverbrauch würde mir auch fehlen, hab ewig in den Objekten danach gesucht. Hätte gedacht, der lässt sich einfach über den SmartMeter abfragen.....

daolis, 2023-10-11:
Da gibt es noch immer da grundlegende Problem dass leider das einzelne lesen der Werte vom inverter relativ lange dauert.
Es gibt ein paar Ideen, aber das muss ich noch testen.
Wenn die werte einigermassen zur selben Zeit kommen, sollte die Berechnung wie in dem oben genannten Forumseintrag beschreiben passen.

gold2605, 2023-10-11:
Hallo @daolis,

der Tagesverbrauch kommt bestimmt direkt vom WR. Ich habe die Situation das dieser nicht mit dem Ertrag in der App-Übersicht übereinstimmt. Dein Wert ist, wie auch direkt unter Gerät -WR in der App ersichtlich , z.B. früh wenn noch kein Ertrag für den Tag vorhanden ist bei ca. 2 kWh, am Ende des Tages fehlen dann aber immer ein paar kWh zu dem Ertrag, den die App direkt in der Übersicht anzeigt. In der App unter Geräte-WR sieht man dazu auch ein Diagramm was zeigt, das bereits während der Nacht Ertrag aufgezeigt wird. 
Das Problem habe ich auch mal zum Support von Huawei gegeben, kam aber nicht viel raus, sie waren der Meinung, dass das Smartmeter falsch angeschlossen ist. Das ist jedoch definitiv nicht der Fall und der Ertrag in der App-Übersicht (Überblick) passt 100% mit den Zählerwerten (Energieversorgunger) überein…..

Hinweis wäre noch, dass in anderen Tools der Ertragswert (js-Skript) mit der Übersicht in der App übereingestimmt hat. Nutzen die dann noch einen anderen Ertragswert für den Tag?

VG

gold2605, 2023-10-12:
Hallo @daolis ,

ich habe noch mal eine Ergänzung zu oben. Heute kamen nur ca. 2kWh vom Dach (ganzen Tag Regen). In der FusionSolar App wird das auch so auf der Übersichtsseite angezeigt. In deinem Adapter werden aber knapp 8KWh als Tagesertrag angegeben, dieser Wert steht auch in der FusionSolar App als Ertrag unter Geräte - WR. In dem dortigen Diagramm sieht man, dass das der Tagesverbrauch ist, den der WR als Energie über den Tag zur Verfügung gestellt hat. Die 8 KWh ergeben sich dann aus den 2kWh die vom Dach kamen und ca. 6KWh aus der Batterie…..Der Wert den der Adapter als PV-Tagesertrag angibt passt bei mir definitiv nicht.

backlink-market, 2023-11-21:
Wie kann denn der aktuelle Hausverbrauchswert aus dem Smart Meter abgefragt werden?
In der App, dass ich gerade ca 1400W verbrauche aber den Wert finde ich nicht in den Objekten.
sun2000-modbus.0.activePower ist viel zu hoch, meist was um die 2000.
Es würde eher passe: sun2000-modbus.0.activePower (2052W)( - sun2000-modbus.0.grid.activePower (722W) solange eingespeist wird. Siehe Screenshot. 

[image]

madmat17, 2023-12-23:
> Wie kann denn der aktuelle Hausverbrauchswert aus dem Smart Meter abgefragt werden? In der App, dass ich gerade ca 1400W verbrauche aber den Wert finde ich nicht in den Objekten. [...]

Eigentlich gar nicht. Der SmartMeter sieht nur, was ins Netz eingespeist wird und was aus dem Netz geholt wird. Der Hausverbrauch muss immer errechnet werden (Summe aus Netzbezug/-einspeisung, PV-Leistung und Akkube-/entladung).

closed 2024-02-02 by daolis