daolis/ioBroker.sun2000-modbus #8: Active Power Einheiten nicht durchgängig

christi-ko, 2023-08-28:
**Describe the bug**  
Bei den States für Active Power wird zwischen W und kW gewechselt.

**To Reproduce**  
Steps to reproduce the behavior:
-

**Expected behavior**  
Alle Active Power Werte als W angeben. Um das Einheiten durcheinander klein zu halten.

**Screenshots & Logfiles**  
[image]

**Versions:**  
 - Version 0.0.1 vom Datum 28.08.2023

**Additional context**  
-

daolis, 2023-08-29:
Hab die Einheiten direkt aus Modbus Interface Definition.
https://javierin.com/wp-content/uploads/sites/2/2021/09/Solar-Inverter-Modbus-Interface-Definitions.pdf

Im FusionSolar Portal werden aber alle in kW angegeben... Sollte wie vermutlich hier auch verwenden... Dann war es nur mehr eine...

christi-ko, 2023-08-29:
In meinem jetzigen NodeRed [https://forum.iobroker.net/topic/51639/howto-huawei-sun2000-wr-modbus-mit-node-red-read-write](https://forum.iobroker.net/topic/51639/howto-huawei-sun2000-wr-modbus-mit-node-red-read-write) und im Energiefluss-Erweitert habe ich gerade Watt,  weill ja die Zahlen nicht so riesig sind. Die letzten Tage meist unter 1kW 😉.

daolis, 2023-08-31:
@christi-ko Was meinst du mit 'habe ich gerade Watt, weil ja die Zahlen nicht so riesig sind.' ?

Ich hoffe nicht dass sich die Einheit auf Watt ändert wenn der wert unter 1000 ist, oder?

Wie oben erwähnt verwende ich die Einheiten aus der Modbus Interface definition.
Wenn wer andere Vorschläge hat, nur her damit.
Wir sollten uns aber asap auf units festlegen, im nachhinein ändern ist dann eher ungünstig.
Wenn scripts erstellt und da werte vergleicht, wir das nach unit änderungen eher nicht mehr so toll funktionieren. ;-)

daolis, 2023-08-31:
Ahh, jetzt hab ich es. Bin wohl anscheinend auf der DC Leitung gestanden ;-)
Also hättest du gerne überall Watt. (Statt kW)

christi-ko, 2023-08-31:
Genau Watt bei der Power und kWh beim den Energiemengen :-) ...

commit 09c241e6 (2023-09-01): Use only Watt, fixed wrong gain for grid P1-3Currents, translations: fixes #14, fixes #8
closed 2023-09-01 by daolis