daolis/ioBroker.sun2000-modbus #16: Einheiten der Watt-Werte teilweise noch falsch

RoHa78, 2023-09-03:
**Describe the bug**  
Umstellung der Werte von KW in Watt sind zwar erfolgt, jedoch sind die Einheitenbezeichner noch in KW (oder ich habe seit Kurzem einen tollen Ertrag ;))

**To Reproduce**  
Steps to reproduce the behavior:
siehe Werte in IOBroker in den Datenpunkten

**Screenshots & Logfiles**  
[image]

**Versions:**  
aktuelle Testversion

daolis, 2023-09-03:
Die States werde zwar erstellt (initial) aber anscheinend nicht upgedated...
Wenn du den State löscht, dann den adapter neu startest, dann sollte es passen.

Aber wie ich sehe hast du für den State schon was konfiguriert. Ich nehme mal an ein speichern in einer DB. Das wäre dann weg und du müsstest es neu konfigurieren...
Oder du stellst es per Hand um: `Edit` -> `Common` -> `Unit`

RoHa78, 2023-09-03:
Dachte die States werden auch automatisch umgestellt. Habe es mal von "Hand" umgestellt, Danke für den Tip.
Seltsam dass bei dem Update dies nicht automatisch aktualisiert wird 🤔

daolis, 2023-09-06:
@RoHa78 Sollte jetzt funktionieren...

RoHa78, 2023-09-06:
Habe die aktuelle Version drüber installiert, die Einheiten sind nun korrekt, tolle Arbeit 😀
Schließe somit diesen Issue!

closed 2023-09-06 by RoHa78