daolis/ioBroker.sun2000-modbus #17: Vorschlag für Update Interval Implementation

christi-ko, 2023-09-10:
**Describe the bug**  
Die aktuelle Implementierung scheint so umgesezt zu sein, dass immer nach dem erfolgreichen ausführen des letzten Intervalls für die die Intervall Zeit gewartet wird. Durch die unterschiedlichen Dauern des Updates ergibt sich daraus einen unterschiedliche Intervalldauer. Eine Synchronisierung mit einer festen Intervalldauer, wie z.B. die von meiner PV Abhängigen Laderegelung ist somit nicht machbar. 
 

**To Reproduce**  
Steps to reproduce the behavior:
Stelle eine `High Intervall` Dauer von 10s ein und schaue auf den Zeitstempel der ÌNFO` im Log.

**Expected behavior**  
Vorschlag wäre die Wiederholung immer zu genannten Zeitpunkt zu triggern, unabhängig von der Dauer. Voraussetzung wäre, dass das Intervall größer der Update Dauer ist.

Die Verzögerung durch das `Low Intervall` könnte man so verbessern, indem man die Werte in kleiner "Happen" aufteilt und z.B. nur 5 Register auf einmal liest. Das könnte man in den dazwischenliegenden "freien" Zeit tun.  Voraussetzung wäre wiederum, dass das `Low Intervall` mindestens 6 x größer als das kleine Intervall ist Um dazwischen die 28 Werte lesen zu können.

**Screenshots & Logfiles**  
[image]

**Versions:**  
Aktuellste Version zum Zeitpunkt.

daolis, 2024-03-04:
Fetchen von Registern habe ich bereits umgebaut. Funktioniert bei mir jetzt wesentlich schneller.
Wird in der nächste Release (in den nächsten Tagen) kommen.

christi-ko, 2024-03-04:
Sehr cool, da bin ich mal gespannt. Blöde Frage nebenbei, das schreiben von Registern  bzw. Einzelwerten geht das auch schon?

daolis, 2024-03-10:
Improved reading of registers from inverter. A bit faster! :wink:

closed 2024-03-10 by daolis