daolis/ioBroker.sun2000-modbus #19: Werte schreiben

gold2605, 2023-10-09:
Hallo @daolis, erst einmal Danke für deine Arbeit, der Adapter funktioniert bei mir schon super zum Auslesen der Werte. Ich bin auf der Suche bzw. es wäre toll, wenn der Adapter auch Werte schreiben könnte. In Verbindung mit einem dynamischen Stromtarif würde ich gern den Akku bei günstigen Strom von AC-Seite laden können. Ich denke, dafür müsste man die TOU Funktion des Akkus nutzen können (eventuell in Verbindung mit der Ladekapazität und SOC des Akkus, zumindest würde das ja jetzt manuell gehen. 
Nach meiner Suche ist das bislang wohl, zumindest im ioBroker, mit keinem Tool möglich ……

viele Grüße

mhonicat, 2023-10-09:
+1 für den Request. Ich würde mir gerne den Smartmeter sparen, die relevanten Daten bekomme ich ja aus meinem EVU Smartmeter ohnehin schon. Jedoch wäre es für den WaF positiv wenn die Huawei App die Daten korrekt anzeigen würde, dafür müsste ich aber Bezug bzw Einspeisung in den WR schreiben.

pluto202, 2023-11-26:
Klingt spannend, würde mich auch interessieren.

daolis, 2023-11-26:
Bin gerade dabei den Adapter in die offiziellen iobroker repos zu bekommnen (Kein install mehr über GitHub notwendig)

Danach mach ich mich an die weitere Entwicklung, und da ist das einer der nächsten auf meiner Liste.

BTW: Adapter testers welcome! https://forum.iobroker.net/topic/70554/test-adapter-huawei-sun2000-inverter

Danny-DFx, 2023-11-27:
> Bin gerade dabei den Adapter in die offiziellen iobroker repos zu bekommnen (Kein install mehr über GitHub notwendig)
>
> Danach mach ich mich an die weitere Entwicklung, und da ist das einer der [...]

Natürlich wird weiter getestet und geholfen. 
Sehr schön. 👍🏻👍🏻

gold2605, 2023-12-23:
Hallo, der Hausverbrauch würde mich auch interessieren (finde den Beitrag leider nicht), hat jemand die Formel (wenn möglich in Verbindung der Objekte) erstellt?

Danke und allen schöne Weihnachten

christi-ko, 2024-03-10:
Ich hänge mich hier auch mal rein. Demnächst hab ich auch Tibber am laufen, das wäre toll wenn das möglich wäre ...

sansonifabio, 2024-05-31:
@daolis bist du noch am Adapter dran? Wollte mal fragen ob das schreiben mal möcglich ist. Siehe bitte meinen Beitrag darüber.

Liebe grüße
Fabio