daolis/ioBroker.sun2000-modbus #20: Sun 2000 Alarmregister abfragen Query alarm register

sansonifabio, 2023-10-09:
Es wäre sehr schön wenn man das Alarmregister abfragen könnte, damit hätte man dann alle wichtigen Daten zu Verfügung wenn was nicht stimmt oder ausgefallen ist, unter anderem auch ob der Strom/das Netz ausgefallen ist.
Herzliche Grüße
Fabio

It would be very nice if you could query the alarm register so that you would have all the important data available if something is wrong or has failed, including whether the power/network has failed.
Best regards
Fabio

[SUN2000 8–28KTL. MODBUS Interface Definitions](https://www-file.huawei.com/~/media/CORPORATE/PDF/FusionSolar/HUAWEI_SUN2000_245KTL28KTL_MODBUS_Interface_Definitions_20150715_JP.pdf)

Liste Modbus Register:

SN Alarm Bit Alarm Name Alarm ID Severity
1 Alarm 1 0 High String Voltage 2001 Major
2 Alarm 1 1 DC Arc Fault 2002 Major
3 Alarm 1 2 String Reversed 2011 Major
4 Alarm 1 3 PV String Backfeed 2012 Warning
5 Alarm 1 4 Abnormal String 2013 Warning
6 Alarm 1 5 AFCI Self-test Fault 2021 Major
7 Alarm 1 6 Short circuit between phase
to PE
2031 Major
8 Alarm 1 7 Power Grid Failure 2032 Major
9 Alarm 1 8 Grid Undervoltage 2033 Major
10 Alarm 1 9 Grid Overvoltage 2034 Major
11 Alarm 1 10 Unbalanced Grid Voltage 2035 Major
12 Alarm 1 11 Grid Overfrequency 2036 Major
13 Alarm 1 12 Grid Underfrequency 2037 Major
14 Alarm 1 13 Grid Frequency Instability 2038 Major
15 Alarm 1 14 Output Overcurrent 2039 Major
16 Alarm 1 15 Large DC of Output current 2040 Major
17 Alarm 2 0 Abnormal Leakage Current 2051 Major
18 Alarm 2 1 Abnormal Ground. 2061 Major
19 Alarm 2 2 Low Insulation Res. 2062 Major
SUN2000MA
MODBUS Interface Definitions 3 Alarms
Issue 01 (2020-06-20) Copyright © Huawei Technologies Co., Ltd. 15
SN Alarm Bit Alarm Name Alarm ID Severity
20 Alarm 2 3 High Temperature 2063 Major
21 Alarm 2 4 Abnormal Equipment 2064 Major
22 Alarm 2 5 Upgrade Failed 2065 Minor
23 Alarm 2 6 License Expired 2066 Warning
24 Alarm 2 7 Abnormal Monitor Unit 61440 Minor
25 Alarm 2 8 Power collector fault 2067 Major
26 Alarm 2 9 Abnormal energy storage
device
2068 Minor
27 Alarm 2 10 Active islanding 2070 Major
28 Alarm 2 11 Passive islanding 2071 Major
29 Alarm 2 12 Transient AC overvoltage 2072 Major
30 Alarm 2 13 External device port short
circuit
2075 Warning
31 Alarm 2 14 Off-grid output overload 2077 Major
32 Alarm 2 15 Abnormal PV module
configuration
2080 Major
33 Alarm 3 0 Optimizer fault 2081 Warning

daolis, 2023-10-09:
There is a more current version available here...
(Solar Inverter Modbus Interface Definitions (V3.0))[https://photomate.zendesk.com/hc/en-gb/article_attachments/4573202911133]

See Alarm Registers on page 11, Reg. 15-17, and Alarm information on page 29.

daolis, 2024-01-11:
@sansonifabio Die AlarmRegister stehen natürlich auch auf der Liste.

Bin mir nur noch nicht ganz sicher wie ich die Einzelnen Werte in ioBroker darstellen soll.
Nur die int Werte sind eher unschön.

Eine Variante wäre ein json State mit allen alarmen die true sind (oder alle mit true/false). Ist nur die Frage wie 'leicht' json dann in diversen Visualisierungen verarbeitet werden kann.

Hast du irgendeine Idee, oder so eine Darstellung in irgendeinem anderen Adapter gesehen?

daolis, 2024-03-04:
Ist teilweise schon implementiert. Wird in der nächsten Release enthalten sein.

sansonifabio, 2024-03-04:
Danke schön @daolis

closed 2024-03-04 by sansonifabio

daolis, 2024-03-10:
Released version 0.1.0.
@sansonifabio Ich bekoommen bei immer genau einen einzigen Alarm.
ID 2266, 'License Expired' (Warning) Bekommt den noch wer?!

sansonifabio, 2024-03-10:
.... gerade keine Zeit teste das nacher.
Grüße
Fabio

sansonifabio, 2024-03-10:
die  0.1.0. vebindet sich nicht mit meinen Huawei Dongle, bleibt gelb und es gibt auch keine Fehlermeldung. 
Bin zurück auf die 0.0.2 da läuft es. Welche Version hat dein Dongle und auch der Wechselrichter.
**Dongel** V200R022C10SPC108
**Wechselrichter** V100R001C00SPC161
**Optimerer** V100R002C10SPC019
**Mbus** V100R001C00SPC330 
**Speicher** V100R002C00SPC127

daolis, 2024-03-11:
Ahh, scheint etwas mit dem connected flag schiefgegangen zu sein.
Ist bei mir auf gelb, aber die Werte werden gelesen. (Nachrichten in den Logs & State updates)

Dongle: V100R001C00SPC159
WR    : V100R001C00SPC159

daolis, 2024-03-11:
Fixed the connection flag issue in 0.1.1

sansonifabio, 2024-03-11:
Guten morgen  @daolis  du  bist ja wieder fleißig, bist du sicher das dein Dongle die 159 hat. Teste gleich die neue Version. Ich hatte gestern nichst in den Logs.

sansonifabio, 2024-03-11:
so läuft soweit:

Bei Datenpunkt:  **Battery temperature** wird angezeigt **335°C** das galube ich eher nicht, da ich dann die Feuerwehr rufen muss.;-)

Bei Alarme habe ich ganz vielen Nullen da stehen **000000000000000000000000000000000000000000000000**
Alarme JSON ist leer denke das ist normal wenn keine Fehler vorliegen.

ansonsten sind wieder Datenpunkte verschoben worden. Kannst du das mal prüfen.

Herzliche grüße
Fabio

sansonifabio, 2024-03-11:
Hello @daolis,

**Version 0.1.2 installed.**
Everything runs like the fire department. No errors discovered so far and none in the log, great job.

Best regards
Fabio

------------------------------------------------------------------------------------------------------------

Hallo @daolis,

**Version 0.1.2 installiert.**
Läuft alles wie die Feuerwehr. Bis jetzt keine Fehler endeckt und auch nichst im Log, Super Job.

Herzliche Grüße
Fabio

daolis, 2024-03-11:
Freut mich zu hören. Die Fetch operationen vom Inverter sollten jetzt um einiges schneller sein.

@sansonifabio Bekommst du auch den Alarm (Licence expired)?
Ich nehme mal an dass es sich um die Lizenz für die 'Intelligente I-V Kennlinie' handelt. Die ich nicht habe...

sansonifabio, 2024-03-11:
@daolis das läuft hier echt schnell, alles super. 
Ich habe nur 00000000000000000 siehe bitte Bild, sind die Nullen normal?
[image]

Sag in den Alarm Json kommen den da Texte an, vertsehe ich das richtig?

daolis, 2024-03-11:
Dann gibt es bei dir keine Warnings, Fehler, etc.
Die 00er sind ok. Alarms beinhaltet die Bitfields von 3 int16 registern = 16 * 3 Stellen lang (Bits als string)
alarmsJSON sieht bei mir z.B. so aus...
```json
[
  {"name":"License Expired","id":2066,"level":2,"levelName":"Warning"}
]
```
Json structure is defined here: https://github.com/daolis/ioBroker.sun2000-modbus/blob/main/src/lib/alarms.ts#L7

sansonifabio, 2024-03-11:
... vielen dank für deine Erklärung.

daolis, 2024-03-11:
Schau ich mir so schnell wie möglich an.
Kennst du zufällig das register, in dem steht dass der WR im Standby modus ist? (Falls es das gibt...)

sansonifabio, 2024-03-11:
32089

sansonifabio, 2024-03-19:
Guten Morgen @daolis,

konntest du schon was machen?
Ich muss den Adapter jetzt immer stoppen wenn der Wechselrichter im Standby ist da sonst mein Log überläuft. 
Wäre schön wenn du das fixen kannst.;-) 
Soll ich ein extra Issisu erstellen?

Grüße
Fabio

daolis, 2024-03-19:
Hi @sansonifabio
Habe fürs erste die Log message auf silly gestellt. Sollte jetzt nicht mehr kommen.
Das eigentliche Auslese passiert immer noch, sollte aber kein problem sein.
Und das setzen der States passiert nicht wenn kein Wert vorhanden ist (log message).

Installier bitte zum probieren die letzte Version vom 'main' branch.

sansonifabio, 2024-03-19:
... vielen Dank @daolis, aber was ist **"Installier bitte zum probieren die letzte Version vom 'main' branch"** ist das von Github?
Grüße
Fabio

sansonifabio, 2024-03-19:
... erledigt sag dir heute Abend Bescheid wenn er ins Standby geht.

sansonifabio, 2024-03-19:
@daolis Wechselrichter ist jetz im Standby aber die Fehler kommen trodzdem noch rein.

sun2000-modbus.0 | 2024-03-19 21:36:47.992 | error | Value for register '32086' is undefined!
sun2000-modbus.0 | 2024-03-19 21:36:14.240 | error | Value for register '32086' is undefined!
sun2000-modbus.0 | 2024-03-19 21:35:40.477 | error | Value for register '32086' is undefined!

daolis, 2024-03-20:
Hmm, sollte nicht so sein. Sieht so aus als ob beim installieren etwas nicht ganz so funktioniert hat.
Die log message sollte auf silly gestellt sein.
siehe https://github.com/daolis/ioBroker.sun2000-modbus/blob/main/src/lib/states.ts#L614C49-L614C67

So sollte es eigentlich funktionieren...
[image]

sansonifabio, 2024-03-20:
.. was soll ich tun @daolis  ich hab es genau so installiert und zwar 3 mal wie du beschrieben hast. 
Evtl. aktualisiert es nicht weil es keine Versionsnummer hast.

sansonifabio, 2024-03-21:
Guten morgen @daolis hab gerade gesehen du hast ne Version veröffentlicht hab sie gerade installiert, sag dir dann heute Abend
[image]
 bescheid.
Vielen Dank 
Herzliche Grüße
Fabio

sansonifabio, 2024-03-21:
Guten Abend @daolis,

der Wechselrichter ist im Standby und der Adapter läuft ohne Fehler und Warnungen. Daten werden weiter abgeholt sieht also super aus. Melde mich morgen früh noch mal um zu sehen ob in der Nacht was aufläuft, was ich aber nicht glaube.

**Vielen Dank ;-)**
Grüße
Fabio

sansonifabio, 2024-03-22:
Guten morgen @daolis,

lief ohne Probleme durch, holt alle Daten, keine Fehler oder Warnungen. 
Kannst du gerne, wenn du Zeit hast ins Stable bringen.;-)

Herzliche Grüße
Fabio