daolis/ioBroker.sun2000-modbus #24: Read data from 2. inverter does not work

TrackerTheCode, 2023-11-07:
**Describe the bug**  
I have connected the first inverter via modbus ID: 1. Unfortunately, the second inverter cannot be connected - Modbus ID: 16.
Additional info from my analyses. I stopped my second instance and change the ID in the first instance to 16. After I successfully saved the new Id, the connection was established and data were displayed, but not correctly. It was exactly the same data as with Id 1. From my point of view, the change of the Id has no impact.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to 'adapter'
2. Click on 'add instnace' two times, to have to instance available
3. configure both adapters, with id 1 and 16

**Expected behavior**  
Second instance of adapter, to connect the other Id should work, so both inverters-instance can be connected and work in parallel.

**Versions:**  
 - Adapter version: 0.0.1
 - JS-Controller version: 5.0.12
 - Node version: v18.18.2
 - Operating system: Docker enviroment on Unraid NAS system

**Additional context**  
I have set up a second instance and taken the ID of the system configuration via the FusionSolar app. The inverters are coupled. The one with ID 1 has the dongle installed.

TrackerTheCode, 2023-11-07:
If i can support with a debug session, i can support in this case.

daolis, 2023-11-14:
@TrackerTheCode 
See #21 Sadly, this will not work (at least for now not with a 2nd instance)

I will check the modbus library if it is easily possible to read modbus registers with a different unitID using the same connection...

Crissiii10, 2024-01-01:
Hi, ich habe das selbe Thema. Ich habe bei mir auch 2 Wechselrichter. Leider werden nur die Daten eines Wechselrichters angezeigt. Falls du etwas benötigst. Melde dich gerne

Crissiii10, 2024-01-01:
> @TrackerTheCode See #21 Sadly, this will not work (at least for now not with a 2nd instance)
> 
> I will check the modbus library if it is easily possible to read modbus registers with a different unitID using the same connection...

Hi, bist du weiter gekommen? Ich habe den Adapter installiert und er läuft soweit auch! Klasse Arbeit! Nur „leider“ habe ich 2 wechselrichter und bekomme nur die Daten von einem angezeigt. Könnte das dass Problem hier sein? 

Danke dir

daolis, 2024-01-03:
I have found something, but it will not work with a 2nd instance (Connected to the same dongle).

I have to change the connection code a bit, and id for the 2nd Inverter will be in config!

Crissiii10, 2024-01-03:
Cool thanks for the Feedback. So you would create a New Version with the New Feature?

Crissiii10, 2024-01-12:
> @daolis Do you have something implemented already. Maybe I can test it for you/support you!?

Sure i have implement it. If you tell me what i should do i can test it

daolis, 2024-01-12:
@TrackerTheCode still work in progress. Need more or less to 're-implement' the communication.

Crissii10, 2024-01-15:
> @TrackerTheCode still work in progress. Need more or less to 're-implement' the communication.
> 
> 

I dont know if it help you but in Home Assistant the Connection to the second work