daolis/ioBroker.sun2000-modbus #25: "daily energy yield" falsche werte im vergleich zu Fusion Solar

Danny-DFx, 2023-11-08:
Hi.

Ich habe festgestellt das im Adapter falsche Tageswerte angezeigt werden:
Die Werte weichen im vergleich zu Fusion Solar doch um einiges ab, ich kann aber nicht erkennen, woran das liegen kann.

Siehe Screenshots im Anhang, die habe ich fast Zeitgleich gemacht.

Viele Grüße
Daniel

[image]
[image]

daolis, 2023-11-09:
Hmm, komisch, der Wert kommt direkt aus dem Wechselrichter.
Möglicherweise kombiniert FusionSolar irgend einen andere Wert damit?!

Danny-DFx, 2023-11-09:
Hi daolis.
Ja, ich verstehe es auch nicht so richtig, da in den offiziellen Dokumenten der richtige Datenpunkt beschrieben steht und in deinem Adapter auch korrekt ausgewählt wird.
Jetzt wird es aber noch merkwürdiger, heute habe ich mir meine Daten von gestern angesehen und Daily Energy Yield in Fusion Solar und in dem von Deinem Adapter abgegriffenen Datenpunkt sind wieder, bis auf wenige Watt, identisch.
Also scheint die Grundsätzliche Abfrage korrekt zu sein, nur die korrekte Darstellung vielleicht einige Zeit verschoben, aber wenn das Live-Werte sind, dürfte es ja eigentlich nicht sein...? 
Ich konnte in den Dokumenten von Huawei aber auch nichts zu dem Thema finden.

Ich schaue mir die Werte von heute gerade mal an... Da zeigt mir Fusion Solar gerade "0" an, während ich in dem Adapter bereits 1,7kWh sehe. Also aktuell genau der umgekehrte Fall zu gestern... 

Von meiner Seite aus, wäre das ok, wenn ich das so weiß und am Ende des Tages die Werte wieder gleich sind, aber interessant wäre es schon zu wissen, warum dieser Unterschied da ist bzw. wie er entsteht.

Grüße
Daniel
[image]
[image]

sansonifabio, 2023-11-09:
Ich kann dir zu Fusion Solar App folgendes sagen das da Zeitunterschiede bis zu 20 Minuten drin sind habe ich bei mir schon festgestellt das hat damit zu tun das es über die Cloud geht. Der Adapter aber holt sich die Daten direkt vom Wechselrichter da kannst du dich aber drauf verlassen. 
Grüße Fabio

daolis, 2023-11-09:
So genau habe ich mir die Werte in FusionSolar noch nicht angesehen.. (20min gap)
Mir ist nur bekannt dass die Werte im FS nur alle 5 min aktualisiert werden.

Aber 'eigentlich' sollte Daily Energy Yield die kWh anzeigen die an dem Tag vom 'Dach' gekommen sind, oder?! (Möglicherweise ein naiver ansatz)
Aber wenn das so ist sollte sich der wert, am Abend, wenn es finster ist, nicht mehr ändern... (theoretisch...)

sansonifabio, 2023-11-09:
Ich denke ja, aber da liegt und hängt es auch an der Cloud. Ich verlasse mich auf die App nicht mehr sie dient für mich nur zur groben Information.

Danny-DFx, 2023-11-09:
So habe ich das auch verstanden, das dieser Wert die Tagesproduktion darstellt.
Interessant ist gerade, die Sonne ist ja mittlerweile untergegangen, das der Wert weiterhin im 0,01kWh Schritt hochzählt, bei jeder Aktualisierung (habe ich auf 60sek. gesetzt).

Danny-DFx, 2023-11-10:
Guten Morgen Zusammen,
Ich glaube es klärt sich auf, den Wert, den ich im FS verglichen habe, ist ein anderer, als der der abgefragt wird. Ich "studiere gerade weiterhin die Anleitungen etc., damit ich den Wert im FS verstehe.

Nehme ich die Seite im FS "Real Time Device Data", sind die abgefragten Werte mit denen im Adapter identisch.
Ich gehe davon aus, das es dann kein Bug ist, sondern sich die Werte im FS anders zusammensetzen. Was am Ende aber noch nicht erklärt, warum der Wert nach Sonnenuntergang immer noch leicht gestiegen ist... ;-)

Von meiner Seite aus, kann der Fall auch wieder geschlossen werden, wenn ich neue Erkenntnisse habe, kann ich sie gern mit euch teilen, oder falls ich sonst helfen kann, gern melden. ;-)

Grüße

daolis, 2023-11-10:
Lassen wir da ticket noch offen. Wenn das ganze geklärt ist können wir es schliessen.

thi-am, 2023-11-13:
Hallo zusammen,
habt ihr auch einen Speicher oder nur die PV?

Ich bekomme vom Adapter für heute 4,09 kW bei dailyEnergieYield angezeigt.
Auf der FusionSolar Website (und App) bekomme ich als Yield Today 0,94 kW angezeigt.
Hat ja auch den ganzen Tag geregnet ;-)
Wenn ich mir dazu noch die Daten vom Speicher anschaue, dann habe ich heute laut FS 0,7 kW geladen und 3,85 kW entladen.
Das hieße doch dann ich habe 0,94 kW Produziert, davon 0,7 kW in den Speicher und 0,24 direkt verbraucht.
Wenn ich jetzt zu den 0,24 kW (direkt Verbrauch) noch die 3,85 kW Verbrauch aus dem Speicher rechne, dann lande ich bei den 4,09 kW die der Adapter anzeigt.

Daher vermute ich, dass die Daten unter dailyEnergieYield das zeigen, was ich aus eigener Produktion (PV+Speicher) verbraucht habe und nich aus dem Netz gezogen habe.
Das würde auch erklären, warum sich die Zahl noch erhöht, obwohl es dunkel ist. Zumindest bei einer Anlage mit Speicher.

Gruß Thilo

daolis, 2023-11-14:
Ich hatte eine kurze Unterhaltung mit dem Erbauer der Anlage. Er meinte auch dass die ganzen Berechnungen mit Akku das ganze viel komplizierter machen.

@thi-am coole Beobachtung! Ich werde es bei mir auch einmal nachrechnen, und dann dir Berechnung dementsprechend anpassen!

St.

Danny-DFx, 2023-11-21:
Hallo Zusammen,
Ja, Akku ist vorhanden.
Wie geschrieben, am Ende des Tages passen die Werte wieder. 
Ich bin nur noch nicht aus den Werten und deren Zusammensetzung im FS 100% schlau geworden. 
Derzeit fehlte mir aber auch etwas die Zeit, mich weiter damit zu beschäftigen. ;-)

Grüße

madmat17, 2023-12-23:
> So genau habe ich mir die Werte in FusionSolar noch nicht angesehen.. (20min gap) Mir ist nur bekannt dass die Werte im FS nur alle 5 min aktualisiert werden.
>
> Aber 'eigentlich' sollte Daily Energy [...]

Huawei subtrahiert in dem in der Cloud gezeigten Wert das, was für den "Betrieb" des Akkus und des Inverters benötigt wird. Ist der Akku auf 0% SoC runter, wird der für den Betrieb benötigte Strom vom Ertrag abgezogen. Dieses Phänomen tritt übrigens nicht auf, wenn man die Entladung auf 5% begrenzt.
Einen 20-Minuten-Gap sehe ich übrigens nicht in der Huawei-App. Lediglich 0-5 Minuten wegen das Update-Invervalls in der Cloud.
Ich denke, dass das die Ursache sein könnte. Oder weichen die Werte bereits Tagsüber eklatant voneinander ab?

Peguschwein, 2024-02-19:
Ich habe erst seit Oktober eine Luna. Vorher die Daten via NodeRed und via APi-Key direkt von Fusion gezogen, da waren beide Werte identisch. Seit die Luna dran hängt sind die Daten Unterschiedlich. Bei Energieertrag steht im Info Menge an PV Storm, die vom Haus und über die Wallbox geladen wird. Von der Anlage ins Stromnetzeingespeiste Strommenge in einem bestimmten Zeitraum.  Beim Verbrauch Menge des Stroms der von der P Anlage geliefert wird einschliesslich der überschüssigen PV Leistung die in Speichergeräten gespeichert wird Und was aus dem Netz gezogen wird. Die Differenz wird noch von den Lade bzw Entladeverlusten herkommt. wird auch noch ca 1KW sein pro Lade bzw Entladezyklus.