daolis/ioBroker.sun2000-modbus #26: SoC translated incorrectly ot "state of capacity"

madmat17, 2023-12-23:
First of all: A VERY promising and cool adapter for ioBroker!
Keep on rocking!

**Describe the bug**  
You translated the register 37760 to "stateOfCapacity", whereas this register is holding the current state of charge.
Register 37760 should be translated to "stateOfCharge".

**Versions:**  
 - Adapter version: 0.0.1

**Additional context**  
The state of capacity would more describe the remaining total capacity compared to the battery's initial capacity (the battery health as known from mobile phones) but that value doesn't seem to be held in an externally accessible register. To my knowledge that value isn't transparent to the user, since the gross capacity and net capacity derive from each other and Huawei / LG already added some "reserves" to cover the wearing of the battery.

daolis, 2024-01-03:
Thanks for reporting it.
The bad think is...
This is more ore less a breaking change... If i change this value, the historical values are stored in ...Capacity, and the noew ones in ...Charge.

madmat17, 2024-01-07:
I know that this might be quite cumbersome, but one option could be to have a translation table for the data object names which the user can configure.
For instance having the adapter writing the discharge power of the battery to the object "dischargePower" per default (like it is right now), but the user could change that one to "Entladeleistung", "ontladingskracht", or "kisütési teljesítmény", ... or whatever object name in ioBroker is convenient for the user.

That would also enable users to maintain their existing database entries (like InfluxDB) in case they already have existing ones with different namings without setting up aliases in ioBroker.

Alternatively the users might set up aliases directly in ioBroker and have them being written to a database (I'm not that firm with ioBroker yet to confirm, whether that'd be working).

daolis, 2024-03-04:
Already implemented. Will come in next release.

commit 3a22f221 (2024-03-10): Changed stateOfCapacity to stateOfCharge, added docs page
closes: #26
closed 2024-03-10 by daolis