daolis/ioBroker.sun2000-modbus #33: Keine Verbindung zumGerät

sansonifabio, 2024-01-31:
**Describe the bug**  
Ich habe mein Bookworm aktualisiert danach bleibt der Adapter gelb. Alle anderenn Adaptert die ich habe arbeiten normal. Ich habe weder die IP oder die ID verändert der Adapter bleibt trotzdem gelb. Das einzige was aufgefallen ist das ich im ioBroker eine Meldung erhalten habe das der Info Adapter veraltet ist und nicht mehr gepflegt wird und Probleme machen könnte. Hat das etwas mit dem Huawei Adapter zu tun, ich denke eher nicht?

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
` sun2000-modbus.0
2024-01-31 14:59:21.615	info	Update initial states

sun2000-modbus.0
2024-01-31 14:59:21.614	info	Create states

sun2000-modbus.0
2024-01-31 14:59:21.553	info	starting. Version 0.0.2 in /opt/iobroker/node_modules/iobroker.sun2000-modbus, node: v18.19.0, js-controller: 5.0.17

sun2000-modbus.0
2024-01-31 14:59:21.447	debug	Plugin sentry Initialize Plugin (enabled=true)

sun2000-modbus.0
2024-01-31 14:59:21.415	debug	States connected to redis: 127.0.0.1:9000

sun2000-modbus.0
2024-01-31 14:59:21.354	debug	States create User PubSub Client

sun2000-modbus.0
2024-01-31 14:59:21.353	debug	States create System PubSub Client

sun2000-modbus.0
2024-01-31 14:59:21.344	debug	Redis States: Use Redis connection: 127.0.0.1:9000

sun2000-modbus.0
2024-01-31 14:59:21.325	debug	Objects connected to redis: 127.0.0.1:9001

sun2000-modbus.0
2024-01-31 14:59:21.323	debug	Objects client initialize lua scripts

sun2000-modbus.0
2024-01-31 14:59:21.290	debug	Objects create User PubSub Client

sun2000-modbus.0
2024-01-31 14:59:21.290	debug	Objects create System PubSub Client

sun2000-modbus.0
2024-01-31 14:59:21.289	debug	Objects client ready ... initialize now

sun2000-modbus.0
2024-01-31 14:59:21.268	debug	Redis Objects: Use Redis connection: 127.0.0.1:9001
host.nuc
2024-01-31 14:59:20.899	info	instance system.adapter.sun2000-modbus.0 started with pid 3735
host.nuc
2024-01-31 14:59:18.370	info	instance system.adapter.sun2000-modbus.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)

sun2000-modbus.0
2024-01-31 14:59:18.301	info	terminating
host.nuc
2024-01-31 14:59:17.838	info	stopInstance system.adapter.sun2000-modbus.0 send kill signal

sun2000-modbus.0
2024-01-31 14:59:17.800	info	Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason

sun2000-modbus.0
2024-01-31 14:59:17.800	debug	Plugin sentry destroyed

sun2000-modbus.0
2024-01-31 14:59:17.799	info	terminating

sun2000-modbus.0
2024-01-31 14:59:17.798	info	Got terminate signal TERMINATE_YOURSELF
host.nuc
2024-01-31 14:59:17.795	info	stopInstance system.adapter.sun2000-modbus.0 (force=false, process=true)

sun2000-modbus.0
2024-01-31 14:59:09.176	info	Update initial states

sun2000-modbus.0
2024-01-31 14:59:09.175	info	Create states

sun2000-modbus.0
2024-01-31 14:59:09.112	info	starting. Version 0.0.2 in /opt/iobroker/node_modules/iobroker.sun2000-modbus, node: v18.19.0, js-controller: 5.0.17

sun2000-modbus.0
2024-01-31 14:59:08.988	debug	Plugin sentry Initialize Plugin (enabled=true)

sun2000-modbus.0
2024-01-31 14:59:08.959	debug	States connected to redis: 127.0.0.1:9000

sun2000-modbus.0
2024-01-31 14:59:08.901	debug	States create User PubSub Client

sun2000-modbus.0
2024-01-31 14:59:08.900	debug	States create System PubSub Client

sun2000-modbus.0
2024-01-31 14:59:08.889	debug	Redis States: Use Redis connection: 127.0.0.1:9000

sun2000-modbus.0
2024-01-31 14:59:08.870	debug	Objects connected to redis: 127.0.0.1:9001

sun2000-modbus.0
2024-01-31 14:59:08.868	debug	Objects client initialize lua scripts

sun2000-modbus.0
2024-01-31 14:59:08.837	debug	Objects create User PubSub Client

sun2000-modbus.0
2024-01-31 14:59:08.836	debug	Objects create System PubSub Client

sun2000-modbus.0
2024-01-31 14:59:08.836	debug	Objects client ready ... initialize now

sun2000-modbus.0
2024-01-31 14:59:08.815	debug	Redis Objects: Use Redis connection: 127.0.0.1:9001
host.nuc
2024-01-31 14:59:08.453	info	instance system.adapter.sun2000-modbus.0 started with pid 3720
host.nuc
2024-01-31 14:59:05.925	info	instance system.adapter.sun2000-modbus.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)

sun2000-modbus.0
2024-01-31 14:59:05.842	info	terminating
host.nuc
2024-01-31 14:59:05.383	info	stopInstance system.adapter.sun2000-modbus.0 send kill signal

sun2000-modbus.0
2024-01-31 14:59:05.340	info	Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason

sun2000-modbus.0
2024-01-31 14:59:05.340	debug	Plugin sentry destroyed

sun2000-modbus.0
2024-01-31 14:59:05.339	info	terminating

sun2000-modbus.0
2024-01-31 14:59:05.335	info	Got terminate signal TERMINATE_YOURSELF`

**Versions:**  
 - Adapter version: <0.0.2>
 - JS-Controller version: <5.0.17> 
 - Node version: <18.19.0> 
 - Operating system: <Debian Bookworm>

**Additional context**  
Add any other context about the problem here.

sansonifabio, 2024-02-01:
Also ich bekomme über telnet kein Kontakt da blinkt nach Eingabe von  telnet 192.168.2.125 502 nur ein curser.
Ich hbae nur für die Optimiere eine neuen Version installiert sonst nix

daolis, 2024-02-01:
Dann scheint keine Verbindung möglich zu sein...
Deaktivier mal die firewall.

Mit `sudo iptables -L` siehst auch die konfigurierten Rules...

sansonifabio, 2024-02-01:
Das ist ein intel nuc nur mit Bookworm mit nix anderem drauf. Kann sich da eine Firewall von alleine installieren?

sansonifabio, 2024-02-01:
Ich habe zum testen jetzt einen anderen Adapter installiert da kommen Daten an aber mit diesen Meldungen:

sun2000.0 | 2024-02-01 15:22:07.405 | info | Connected Modbus IP to: 192.168.2.125 /PORT 502
-- | -- | -- | --
sun2000.0 | 2024-02-01 15:22:06.403 | info | Open Connection...
sun2000.0 | 2024-02-01 15:22:06.402 | warn | Error while reading from 192.168.2.125: [32064\|2] id: 1 '' with : Timed out
sun2000.0 | 2024-02-01 15:21:08.685 | info | Connected Modbus IP to: 192.168.2.125 /PORT 502
sun2000.0 | 2024-02-01 15:21:07.682 | info | Open Connection...

daolis, 2024-02-02:
Das scheint genau das selbe Problem zu sein. Er versucht die Connection zu öffnen bekommt aber einen timeout.
Modbus Einstellungen im Inverter. Da kann man den Zugriff so weit ich weiss auf gewisse IPs geschränken. So wie eine Allowlist.

daolis, 2024-02-02:
Schau dir mal die Konfiguration von deinem Dongle im FusionSolar an.
Gerätemanager -> Dongle -> Konfiguration
Das sollte auf uneingeschränkt sein...
[image]

sansonifabio, 2024-02-02:
Guten morgen @daolis wenn der Elektriker da war lasse ich die IP vom Iobroker eintragen dann sehen wir weiter. Du denkst also auch das es am Wechselrichtert liegt? 
So die Einstellung ist so wie bei dir. Sag mal hat dein Dongel auch die IP 100?
Grüße
Fabio
[image]

daolis, 2024-02-02:
IP 100?
Ich hab den Dongle bei mit per LAN Kabel angeschlossen, und am DHCP Server für den Wechselrichter eine Fixe IP eingtragen.
Ich verwende aber einen 10.1.2.* range.

daolis, 2024-02-02:
@sansonifabio 
UPDATE: Check tool
https://github.com/daolis/huawei-sun2000-check
Du kannst die Binaries hier herunterladen...
https://github.com/daolis/huawei-sun2000-check/releases

sansonifabio, 2024-02-02:
Hallo @daolis  keine Ahnung was du meinst mit deinem Tool da reicht mein Wissen nicht für. 
Mein Dongel ist auch im Lan und hat von der Fritzbox ne feste IP zuweisung. 
Entschuldige das soll heißen ID 100 und nicht ip.

daolis, 2024-02-02:
Ahh, du meinst die Modbus UnitID.
Nein. Meiner hatte von Anfang die ID 1

daolis, 2024-02-02:
Möglich!

sansonifabio, 2024-02-02:
[image]
funktioniert irgend wie nicht er sagt immer ich soll cmd öffnen wenn ich dan den Befehl eingebe kommt die Meldung befehl falsch

daolis, 2024-02-02:
Ahh, sorry. `./` muss man unter Linux verwenden, weil man per default keine binaries aus dem Aktuellen Verzeichnis starten kann.

Unter WIN wäre es: `huawei-sun2000-check.bat --ip ...`
[image]

sansonifabio, 2024-02-02:
@daolis **huawei-sun2000-check.bat --192.168.2.125** funktioniert auch nicht. Bin ich da zum dumm für?

daolis, 2024-02-02:
Wenn die timeout message kommt, dann lässt dich der WR nicht... (oder irgendetwas dazwischen)

sansonifabio, 2024-02-02:
..... @daolis was bleibt da jetzt noch fritzbox neu starten und oder Adapter deininstallieren und dann neu installieren?

sansonifabio, 2024-02-02:
@daolis  vielen Dank für deine Hilfe. Der Elektriker versucht nächste Woche vorbeizukommen da er diese Einstellungen nur direckt mit Handy am Wechselrichter ändern kann. Von der Ferne her konnte er keine Fehler oder falsche Einstellungen erkennen. Ich denke auch das beim Update der 15 Optimiere was falsch gelaufen ist. Evtl. läßt mich das Dongel nicht rein das könnte doch auch eine Möglichkeit sein.

sansonifabio, 2024-02-05:
Guten morgen @daolis ja was soll ich sagen, ich habe die Tage viele getestet nichts kam dabei raus. Habe verschiedene Adapter probiert, wie auch Software da kamen die Verbindugen zu stande **aber die Werte kamen eher unzuverlässig und einige gar nicht**.  Dongel, Wechselrichter neu gestartet im ioBroker den Adapter gelöscht und neue installiert. Nix brachte eine Änderung.
Eines hat aber alles gemeinsam das die IP wie auch der Port und die ID immer noch die selben waren und folgerichtig auch funktionieren. 

Heute morgen habe ich dann aus Frust den Adapter wieder mal neu gestartet und was soll ich sagen er wurde grün und es kamen Daten an, bis jetzt soweit alle, wenn der Wechselrichter in Betrieb geht schau ich ob dann alle Daten aktualisiert werden noch ist er im Standby. Ich habe den Adapter jetzt mehrmals neu gestartet und auch die Einstellung der Intervalle geändert um zu sehen ob es daran gelegen hat. Spielt aber keine Rolle er wurde immer wieder grün. Alles sehr sehr komisch. Ich habe jetzt auch den Support von Huawei angeschrieben mal sehen ob die was feststellen konnten. 
Ich melde mich heute Abend dann noch mal. Danke dir @daolis 

Herzliche Grüße aus Köln
Fabio

daolis, 2024-02-06:
@daolis Okeee. In perfekt English: I understand only railway station! :wink:
Das ist allerdings ein sehr eigenartiges Verhalten...

sansonifabio, 2024-02-06:
@daolis ja was soll ich sagen. 
Es läuft. 
Ich glaube das es der Dongle war, aber wie beweisen? Der hat mir eigentlich viel zu oft, zu lang hintereinander grün geblinkt. Huawei hat sich gemeldet das sich das einer ansehen wird. 
Was sagst du @daolis  sollen wir hier jetzt zu machen?

Herzliche Grüße
Fabio

sansonifabio, 2024-02-08:
Hallo @daolis  Huawei hat sich gemeldet und gesagt das ich meine MBUS Version updaten muss. 
Da ich keinen Zugriff uaf Updates habe, macht das demnächst mein Elektroinstallateur. 
Ob das im Zusammenhang steht mit dem Modbus, kann ich nicht sagen, es läuft ja alles deshalb mach ich hier zu.

Herzliche Grüße
Fabio

closed 2024-02-08 by sansonifabio

daolis, 2024-02-08:
@sansonifabio Könntest du bitte noch die Version auf die du updaten sollst hier posten?
Würde sie gerne mit meiner vergleichen...
Danke!

sansonifabio, 2024-02-08:
@daolis klar, das ist die Version V100R001C00SPC330 aber Achtung wen du/ihr das updatet solltet Ihr auch euer System mit updaten hier ein Link zu der Seite wo ihr genau sehen könnt was aktuell für Huawei PV und Zubehör ist: 
https://photomate.zendesk.com/hc/en-gb/categories/4406398674589
Und noch was die Optimierer und das MBUS nur aktualisieren bei Sonnenschein sprich wenn die Anlage in Betrieb ist und dei Panels Online.

Grüße
Fabio