daolis/ioBroker.sun2000-modbus #34: Weitere Register abfragen

Basti-RX, 2024-02-07:
Hallo,
wäre es möglich weitere Register abzufragen?
Folgende Register wären aus meiner Sicht zur auswerten noch gut.

[image]
[image]

Basti-RX, 2024-02-19:
Hi. Ich habe jetzt schon einige Pdf's durch was Modbus bei Huawei angeht, aber die Adressen kenne ich nicht. Hast du eine Quelle dafür?

Peguschwein, 2024-02-19:
[image]

[image]

kommt aus dem Modbus interface Definitions  Issue 4 vom 29.3.22 Gibt unterdessen ein neueres Dokument aber die Adressen sind sicher noch die gleichen

Basti-RX, 2024-02-19:
Kannst du die im gesamten hochladen?

Peguschwein, 2024-02-19:
[Solar Inverter Modbus Interface Definitions 5.pdf](https://github.com/daolis/ioBroker.sun2000-modbus/files/14336439/Solar.Inverter.Modbus.Interface.Definitions.5.pdf)

Hier das gesammte Dokument. Habe da so einiges gesammelt da ich über 1 Monat mit dem Huawei Support dran war

Basti-RX, 2024-02-19:
Danke. Ist eine neuere Version als die ich kenne. Lese mir die mal durch und dann gucken wir mal was @daolis dazu meint.

Basti-RX, 2024-02-19:
Wallbox ist bei mir leider nicht vorhanden . Ansonsten sind WR und Batterie gleich.

Basti-RX, 2024-02-22:
@Peguschwein guckt dir den adapter an. scheint gleich zu sein aber besser weiter entwickelt. kann man auch schon offiziell wohl im iobroker beziehen ohne git

Peguschwein, 2024-02-22:
@Basti-RX ja habe es soeben installiert sieht gut aus. Grr.... habe ich vergebens ein Bloxkly für die Stringleistung gemacht. Ist ja auch schon bereits drin. Wenn ich noch einen Wunsch anbringen darf String auf 6 odre 8 erhöhen. Mein Kollege hat 42kwp auf dem Dach da reichen 2 Strings nicht mehr :-)

Peguschwein, 2024-02-23:
Ich denke man sollte die offizielle ModbusAdresse imNamen drin haben. Bei soviel Infospunkte verliert ma schnell den Überblick. z.B 37780_Total _Charge. Ichb bi die neuen Datenpunkte am zuteilen und habe da so meine Mühe dn richtigen zu finden. In der Modbusdefinition ist schön nach Gerät, Adresse und Name. Würe das so übernehmen.

daolis, 2024-02-26:
> Danke. Ist eine neuere Version als die ich kenne. Lese mir die mal durch und dann gucken wir mal was @daolis dazu meint.

Habe auch den Support kontaktiert, umd die neueste Version zu erhalten. Habe leider auf der Huawei Seite nichts gefunden....
Mal sehen was ich da bekomme! :wink: 

Aber danke fürs hockladen der 5er Version, die kannte ich noch nicht...

daolis, 2024-02-26:
> Ich denke man sollte die offizielle ModbusAdresse imNamen drin haben. Bei soviel Infospunkte verliert ma schnell den Überblick. z.B 37780_Total _Charge. Ichb bi die neuen Datenpunkte am zuteilen und [...]

Gute Idee, was haltet ihr davon es in die Description zu schreiben? (siehen Screenshot)

[image]

sansonifabio, 2024-02-26:
@daolis ...finde ich gut.

daolis, 2024-03-10:
most of them in version 0.1.0

closed 2024-03-10 by daolis