daolis/ioBroker.sun2000-modbus #36: Add Huawai Smart Charger

hpoeppel, 2024-02-10:
Es wäre super wenn wir auch noch die Huawai Wallbox auslesen könnten:
[image]

Ich denke man muss dann aber zusätzlich die IP-Adresse der Wallbox hinterlegen.

Ich stehe gerne für Tests zur Verfügung ;-)

Huawai Wechselrichter, mit Akku, Wallbox und E-Auto sind vorhanden.

Peguschwein, 2024-02-19:
[image]
[image]

An der Wallbox Problematik bin ich schon eine Weile dran. Die Grafik wie von Dir eingestellt habe ich via Support erhalten. Aber sonst gleich nichts. Ich denke aber nicht das die Daten via Dongle ausgelesen werden können. Sonst wären sie bestimmt auch über das Fusion Solar Portal auch zur Verfügung. In der App gibt es einen menpunkt Drittanbieter Managementsystem. Dahinter kann eine Modbus Adresse eingegeben werden und die Portnummer. Diese müsste sicherlich nicht eingegeben werden wenn die Komunikation via Dongle funktionieren würde. Haben wir hier keinen Modbus Spezi ?  Braucht es ev einen zusätzlichen Modbus Master oder Modbus Client ? Kenn mich da zuwenig aus

hpoeppel, 2024-04-29:
Ich habe mir noch ein wenig den Netzwerkverkehr zwischen Wallox und Wechselrichter angeguckt,
Die Wallbox spricht immer nur den Wechselrichter per Modbus an und ließt beim Wechselrichter mehrere Register aus. 
Das heißt die Wallbox steuert sich anhand der Register im Wechselrichter selbst.
Die Register die im Wechselrichter ausgelesen werden sind auch immer die gleichen. Das auslesen erfolgt im Sekunden Rythmus:
[image]

Ich mache bei Huawai nochmal nen Ticket auf ;-)

daolis, 2024-04-29:
Das sind exakt die Register des 'Meters'.

[image]
[image]

bernhard2901, 2024-06-01:
Hallo, auch von mir erstmal vielen Dank für den Adapter und deine Bemühungen.
An der Integration der Walbox wäre ich natürlich auch sehr interesiert.
Der Adapter sonst läuft bei mir bis jetzt problemlos...