daolis/ioBroker.sun2000-modbus #42: Slave device failure (device reports internal error)

sansonifabio, 2024-04-14:
**Describe the bug** 

Hello @daolis, I have had this message as a warning in ioBroker since yesterday.
What has changed is that we have upgraded our storage from 5 kwh to 10 kwh.
Then the inverter went to V162 and the dongle to V114.
Otherwise the adapter will continue to run like the fire department.

Best regards
Fabio

Hallo @daolis ich habe seit gestern diese Meldung als Warnung im ioBroker.
Geändert hat sich das wir unseren Speicher von 5 kwh auf 10 kwh aufgerüstet haben. 
Dann der Wechselrichter auf V162 gehoben haben und den den Dongel auf V114. 
Sonst läuft der Adapter weiter wie die Feuerwehr.

Herzliche Grüße
Fabio

**Screenshots & Logfiles**  

sun2000-modbus.0 | 2024-04-14 08:35:17.913 | warn | Error while reading block from 192.168.2.125: [32064-32082] '' with : Error: Modbus exception 4: Slave device failure (device reports internal error)

**Versions:**  
 - Adapter version: 0.1.3
 - JS-Controller version: 5.0.19
 - Node version: 18.20.2
  -NPM: 10.5.0
 - Operating system: Debian Bookworm

TobiasF1986, 2024-04-22:
Hello,
I also have this warning.
I got this messages while using the sun2000 adapter from bolly. Maybe it isn't an issue from the adapters themselves but from our systems?
I use a SUN2000-10KTL-M1 with dongle, an two packages LUNA2000-5KW.

"Error while reading from 10.217.222.176 [Reg: 37765, Len: 2, modbusID: 1] with: Modbus exception 4: Slave device failure (device reports internal error)"
"Error while reading from 10.217.222.176 [Reg: 32080, Len: 2, modbusID: 1] with: Modbus exception 4: Slave device failure (device reports internal error)"

daolis, 2024-04-22:
Hi @sansonifabio, @TobiasF1986,

My versions:
Dongle: V200R022C10SPC108
Inverter: V100R001C00SPC159

I do not get any error/warning.
Do you see any alarm/warning in fusionSolar?!

St.

TobiasF1986, 2024-04-22:
No, in fusionsolar everything works fine without any alarms.

sansonifabio, 2024-04-22:
@daolis 
Nein bei mir auch kein Fehler in der App uund Fusion Solar die Warnung tritt sporadisch auf man kann sie nicht an was bestimmtes fest machen auch dein Adpter steigt da nicht aus. 
Hat es etwas mit der Erweitrungen vom Speicher zu tun? Und was sind das den für Register weißt du das 32064-32082?

Grüße
Fabio

@TobiasF1986  **zwei Pakete LUNA2000-5KW** hattest du die von Anfang an, oder hast du 5 kwh kürzlich nachgerüstet.
liste mal bitte dein Firmware auf und wie ist die Zustandsprüfung der Battereien?

Grüße
Fabio

daolis, 2024-04-22:
@TobiasF1986 Which adapter version are you using?
Because in the newest you shoud see `Error while reading block from...`.
This version is much faster, but anyways, it's happening in the newest, too.

But for me it looks like, theres a bug in the newest firmware. As mentioned, runs without errors with older firmware.

Another question: @sansonifabio @TobiasF1986 
Is every read failing, or does it happens sometimes?!
(Check the state update time for 32064 [inputPower], 32078 [peakActivePowerCurrenDay])

These 2 registers are read in one block with one read operation.

TobiasF1986, 2024-04-22:
Die Anlage steht von Anfang an so da. 
Dongle-HV2310510907 mit V200R022C10SPC114 
SUN2000-10KTL-M1 mit V100R001C00SPC162
Meter-1 Power Sensor Power Sensor(Export+import meter)  (keine weiteren Informationen in fusion solar)
LUNA2000-5KW-C0 Battery Pack1 und Pack2 mit jeweils V100R002C00SPC127
"Battery Health Check pending"

sansonifabio, 2024-04-22:
@daolis **(Überprüfen Sie die Statusaktualisierungszeit für 32064 [inputPower], 32078 [peakActivePowerCurrenDay])**

die werden bei mir normal aktualisiert keine Besonderheiten. Der Fehler, hatte ich ja schon geschrieben kommt sporadisch und man kann ihn nicht an eine bestimmte Situation festmachen.

@TobiasF1986 also wir haben eine ähnliche Anlage okay, jetzt die Frage wann trat der Fehler zuerst bei dir auf nachdem du das Dongel auf 114 gehoben hast oder den Wechselrichter auf 162?

daolis, 2024-04-22:
Bei ist die Anlage auch ähnlich..
SUN2000-10KTL-M1 wit 10kWh Speicher. (Nur nicht ganz up to date...)

sansonifabio, 2024-04-22:
> Bei ist die Anlage auch ähnlich.. SUN2000-10KTL-M1 wit 10kWh Speicher. (Nur nicht ganz up to date...)

Dann bring sie doch mal auf den neusten Stand wenn du Lust hast, dann sehen wir ja ob es dann von den neuen Firmware Versionen von Huawei kommt.

Grüße
Fabio

sansonifabio, 2024-04-22:
... okay dann hat es wohl damit nix zu tun. 

Bei mir trat der Fehler erst auf nacdem ich mein System auf den neusten Stand gebracht habe. Und zwar alles auch MBUS und  Optimiere etc.

sansonifabio, 2024-04-28:
Unfortunately I can't tell you anything new,
The error occurs sometimes and then not again. It cannot be tied to any event.

Leider kannich euch nix neues berichten,
der Fehler tritt mal auf und dann wieder nicht. Man kann ihn an kein Ereigniss festmachen.

Grüße
Fabio

sansonifabio, 2024-05-21:
@daolis also ich konnte den Fehler lokalisieren. Der Adapter verliert für ca. 4 -5 sec. die Verbindung zum Wechselrichter warurm das so ist weiß ich leider nicht. 
Tatsache ist das der Dongel online ist und die PV Analge läuft. 
Evtl. kann man das im Adapter so abfangen das es erst zu einer Warnung kommt wenn der Adapter nach einer bestimmten Zeit keine Verbindung bekommt.

Grüße
Fabio

sansonifabio, 2024-05-31:
@daolis I'm now using the Version Inverter on V100R001C00SPC159. I don't know why the higher one was taken back. 
But the error hasn't occurred for 48 hours now. I think it was due to the firmware configuration of the components.
I'll close it here then. If anything serious happens, I'll reopen.

Greetings
Fabio

@daolis bin jetzt mit der Wechselrichter Version auf V100R001C00SPC159 warum man die höhere zurückgenomen hat weiß ich nicht. 
Aber der Fehler trat jetzt 48 Stunden nicht mehr auf. Ich denke es lag an der Firmware Konstelation der Komponeten.
Ich mache dann mal hier zu. Falls was gravierendes sein sollte öffne ich wieder.

Grüße
Fabio

daolis, 2024-05-31:
Hi @sansonifabio,

Danke für die Info. Ich hatte vor ein paar Tagen einen eigenartigen Fall (FW: ...159).

Von einem Moment auf den anderen sind keine Daten mehr übertragen worden.
Adapter restart, etc. hat nicht geholfen.
Dann einmal mit Installateu-Account angemeldet (nichts geändert) und es hat wieder funktioniert.

Nur zur Info, falls es bei dir auch mal auftreten sollt. Sieht so aus als ob sich noch 1, 2, n Bugs in der FW verstecken... 

lg
  St.

sansonifabio, 2024-05-31:
@ ich lache mich weg, eben schau ich ins Log vom ioBroker. Zwei Tage nichst udn jetzt doch wieder ;-)))

sun2000-modbus.0 | 2024-05-31 06:59:33.359 | warn | Error while reading block from 192.168.2.125: [32064-32082] '' with : Error: Modbus exception 4: Slave device failure (device reports internal error)

sansonifabio, 2024-06-21:
@daolis ... nix neues, der Fehler tritt weiterhin sporadisch auf und läßt sich an nichts festmachen. 
Der Adapter läuft aber weiter ohne Probleme ich denke das es was mit dem Dongel zu tun hat und den Webportal von Fusion Solar evtl. kommt sich da was in die Quere. Ist nur eine Vermutung.

Herzliche Grüße
Fabio

sansonifabio, 2024-07-06:
@daolis so habe jetzt volgendes gemacht, Dongle abgezogen 2 Minuten liegen lassen, dann wieder anklemmen und wenn alles grün ist Wechselrichter neu gestartet sprich 1 Minute vom Netz genommen.
Seit einer Woche habe ich den Fehler nicht mehr. Ich mache jetz zu. ;-)

Herzliche Grüße
Fabio

closed 2024-07-06 by sansonifabio