daolis/ioBroker.sun2000-modbus #47: Gesamtverbrauch Datenpunkt hinzufügen

sansonifabio, 2024-10-01:
Hallo @daolis,

sag mal ist es möglich das du bitte noch den Datenpunkt Gesamtverbrauch einfügen kannst. Also die Summe was vom Netz und PV verbraucht wurde?

Herzliche Grüße
Fabio
PS: **Sag mal bist du noch aktiv?**

daolis, 2024-10-02:
Hi @sansonifabio ,
Hatte leider nicht viel Zeit daran weiterzuarbeiten.

Was verstehst du genau unter Gesamtverbrauch?
* Den aktuellen (gerade jetzt), nur intern, oder mit einspeisen?
  * Das wäre dann eine Berechnung des aktuellen Verbrauchs von PV + Verbrauch von Netz.
* Wie viel aktuell Verbraucht WURDE (pro Tag, Monat, Jahr)
  * Wird schwieriger, da ich die SUMMOIERTEN Werte nicht im WR gefunden habe
  * Es gibt bei Grid `reverseActivePower`, die gibt aber an wie viel gesamt (von Inbetriebnahme der Anlage) vom Netz gezogen wurde.

sansonifabio, 2024-10-02:
Hallo @daolis, 
schön von dir zu hören.;-)
Ich meine den Netzbezug plus wie auch den verbrauchten PV Strom, schau mal bitte in der Fusion Solar App da wird es angezeigt. Evtl. wird das aber nicht im Wechselrichter abgelegt sondern in der Cloud berechnet, das weiß ich aber nicht.

Herzliche Grüße
Fabio

sansonifabio, 2025-01-30:
@daolis hallo, bist du noch aktiv?

Hezliche Grüße
Fabio

sansonifabio, 2025-03-15:
... da hier anscheinend nix meer passiert und man auch keine Antworten mehr bekommt. Mach ich jetzt hier zu.

closed 2025-03-15 by sansonifabio